KONNEKTING/KonnektingDeviceLibrary #20: DPT16 and DPT232

ouranos6, 2016-08-26:
Hi,

Sorry for the duplicata but I'm not already on the move with the forum, Github etc... but I want to help 

I've added the dpt 16 and 232 to use the strings (14char) and the 232 for the RGB I found the information here
https://knx-user-forum.de/filedata/fetch?id=705672

Let me know if it's good or not and if it needs more work to use theses strings and rgb colors.

BR,

Ivan
[KnxDPT.h.txt](https://github.com/KONNEKTING/KonnektingDeviceLibrary/files/438688/KnxDPT.h.txt)

tuxedo0801, 2016-09-14:
Code added by @Eugenius with commit dc0df99c6101195e0412056cb1ad632feeda138e.

One has to use the read/write methods with byte[] argument, as there is no convenient method to read/write those types. See KNX DPT spec for more details about the DPT formats.

tuxedo0801, 2016-09-14:
DPT232.600

Carries 3 bytes. So use byte[3] array, where
byte[0] = B
byte[1] = G
byte[2] = R

tuxedo0801, 2016-09-29:
Works.

closed 2016-09-29 by tuxedo0801