KONNEKTING/KonnektingDeviceLibrary #38: Use interrupt based telegram reception

tuxedo0801, 2016-10-21:
Until now, the lib requires the user to call Knx.Task() in the loop() function. To not miss any bytes/telegrams, the Knx.Task() function has to be called every 400µs.

That makes things complicated, especially when you try to use 1wire or I2C, where reading data can cause several ms of delay in loop()...

-> Investigate how interrupt can be used with UART on arduino plattform (32u4 and SAMD21G18 ...)
-> Check the required code changes, especially wrt. buffering incoming data before data is parsed/evaluated

tuxedo0801, 2016-10-24:
See http://forum.arduino.cc/index.php?topic=430894.0

After some internal discussion, we will try the following with beta5:

* Knx.Task() will be timer-driven. 0,5ms to 1ms... check what works best
* Libraries may disable interrupts (see: https://www.arduino.cc/en/Reference/NoInterrupts), but no longer than the time one single byte needs to receive (19200Baud --> ~520µs)

--> It's still inclear what "noInterrups" deactivates. _all_?? If serial IRQ will be still alive, libraries max disable interrupts longer (about 33ms --> 19200baud + 64byte buffer = ~33ms).

tuxedo0801, 2016-10-24:
As far as I found out, _all_ is deactivated (counts for AVR and SAMD21). So libraries have to enable interrupts latest ~400..500µs after it get's disabled.

Adminius, 2017-04-20:
can we stay in Knx.Task() until we've recived a full telegramm? => Knx.Task() blocks loop().

For that case if it is possible to have serial interrupt driven function call (e.g. SAMD21 hast up 4x "free" and undefined serials, that can be used for that), should we call Knx.Task() or another "special" function?

tuxedo0801, 2017-04-20:
For now, I would not change anything. There's enough RX Buffer to be safe for ~2-3sec or even more (depending on load on the bus) in a normale one-family-house installation.
The above mentioned ~33ms is valid for 100% bus load, which is in most installations not the case. 

Adding an irq-handler for an additional serial port is possible, but it's more like a "special hack" for SAMD. 

If one has some fear that we miss sime packets: Use a timer which triggers within ~33ms at least once. 
That solves the issue, as long as there is no conflict with other libs that also work with an interrupt-handler.

Using a timer is in the end exactly the same as having an interrupt driven bcu read....

tuxedo0801, 2017-06-06:
maybe obsolete due to #43

tuxedo0801, 2018-01-11:
Todo:

Create header-file containing the timer-implementation + timer-init-function.
Result: The user can include the header-file + call timer-init-function in setup()

Question: Which interval?

<2,5ms until we remove the timeout-based-telegram-end-detection
~1sec if we remove the timeout-based-telegram-end-detection

Adminius, 2018-08-12:
currently timer based reception don't work at all.
i don't know the reason. May be some variable shold be voilatale may be some function should be rework... we should debug it or complety rewrite...