KONNEKTING/KonnektingDeviceLibrary #39: only INTx Pin for programming input

mrniemand, 2016-11-05:
In the moment only an INTx Pin can be used as input for the programming button.
This is caused by KnxTools::init
attachInterrupt(digitalPinToInterrupt(_progButton), KnxToolsProgButtonPressed, RISING);
an more generic version, which uses an pcint interrupt should be better. (Intx is free for more complex tasks, and layouting may be easier)
In the moment I use an complete PCINT Library for this, so the line changes to attachPinChangeInterrupt(digitalPinToPinChangeInterrupt(_progButton), KnxToolsProgButtonPressed, RISING); 
so that every pcint pin can be used. But I don't think using this external library "PinChangeInterrupt" for this is an great idea, because I'm afraid that there are hundrets of versions out there.

tuxedo0801, 2016-11-05:
Instead of changing the library to use PinChangeInterrupts, why not change your sketch to use PinChangeInterrupts? That keeps the library as universal (and small) as possible (with least dependencies) and if you need more interrupt pins, use PCint in your sketch.

mrniemand, 2016-11-05:
also a good solution, but then I suggest to do the attachInterrupt only when progButtonPin>=0 in the init function. Then I can work with the library as it is.

tuxedo0801, 2016-11-05:
PCInt is not portable enough to integrate it. Also I would stick to an real interrupt for the prog button.

If you need more Interrupts, use the pcint. If this conflicts with the prog button pin, you are free to work around this issue.
We can give some hints how to do this if you Provider a practical example.

mrniemand, 2016-11-05:
yes I understood that PCInt is not portable enough. Because of that I suggest the following:
in KnxTools::init
if (_progButton>=0) attachInterrupt(digitalPinToInterrupt(_progButton), KnxToolsProgButtonPressed, RISING);

I externally call the function Tools.toggleProgState() from my sketch when an pcint happens

tuxedo0801, 2016-11-05:
The method is not meant dir being called externally and migth also change in future. I don't see why the lib should give away controller over prog mode. There's a giod reason why we stick to an real interrupt.

If the progbutton pin conflicts with your pxintt usage, Just exclude this pin from change check.

mrniemand, 2016-11-05:
I prefered an PCInt pin because it made my pcb design a little bit easier. Because of that my prog button is wired to an "normal" io pin, not to an INTx capable pin.

Adminius, 2017-04-19:
I have simmilar issue:
it should be possible to active ProgMode by "double clockling" or "log press" on the button. So the same button can be used user input and ProgMode.
In other cases i should be possible to active ProgMode by "combinations" e.g. TwoButtonsClick or so.

That means: Expert users needs an official way to enable ProgMode without ProgButton definition (and without interrupt)

tuxedo0801, 2017-06-06:
Current idea:

Provide additional init() variant which does not take an pin-number but a function-pointer etc...
So everyone can handle it diffently and code is still backwards compatible.

tuxedo0801, 2018-01-11:
Function-Ptr is not possible, because Lib will not call a function, but a function will tell the lib about the prog-mode.

Adminius, 2018-05-17:
this feature is implemented in beta4a.
You can check the example:
[https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/examples/DemoSketch_without_pins/DemoSketch_without_pins.ino#L132](https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/examples/DemoSketch_without_pins/DemoSketch_without_pins.ino#L132)

closed 2018-05-17 by Adminius