KONNEKTING/KonnektingDeviceLibrary #40: KnxTools::reboot() on Mega644pa

mrniemand, 2016-11-05:
The mega644pa seems to have the same issue like the 328p.
So this type also needs the asm volatile ("  jmp 0"); for rebooting.

Is there any adavantage for the watchdog version instead of the jmp 0?

tuxedo0801, 2016-11-05:
mega644 ist not supported :-(

Supported µC:

* 328p
* 32u4
* esp8266
* samd21

tuxedo0801, 2016-11-07:
See also:

tuxedo0801, 2017-04-20:
We will add a flag that you have to set via `#define ...` to force the asm-jmp-reboot.

tuxedo0801, 2018-01-11:
#ifdef starts with all supported µC, and last #else will use soft reset. ---> All unsupoprted µC will then use soft reset.
See: https://github.com/KONNEKTING/KonnektingDeviceLibrary/pull/44/commits/f32d9c3b693c6a02698aa89f191db72e49c81e59 + swap last wdt with soft reset

Adminius, 2018-05-17:
beta4a was released. we are now using asm-reset for all AVR´s.
@mrniemand pls check if it solves your issue

Adminius, 2018-06-10:
feature seems to work. i will close this issue. 
in case of problems, pls reopen it.

closed 2018-06-10 by Adminius