KONNEKTING/KonnektingDeviceLibrary #41: DeviceLib is hanging if an adress with 1 as last adress part is used

hasselk, 2016-12-11:
please try to reproduce. Im using BETA4 of the Library.

If i program in Adress with 1 as last part like 2/0/1 or 3/0/1 the device stops responding to further programming attempts even if you press the prog Button there is no reaction anymore.

i have to clean the eeprom and reflash the konnekting sketch and then it works again

tuxedo0801, 2016-12-12:
Device does not "hang" if you use different address?

Please attach the serial port, compile sketch with debug and capture serial console output for device startup (after eeprom clean) + complete programming process. Then attach the log to this issue.

hasselk, 2016-12-12:
No any other tested adress works. Tried different combinations and only if 1 is the last part of an adress this happens. Retried with freshly installed Beta4 Lib and the KONNEKTING DemoSketch.
anything works until you give it an adress like 1/1/1, this seams to trigger an endles reset loop

`
--------------------------------------------
- programming with PA 1/1/2 which works
---------------------------------------------------
konnektingKnxEvents index=255
internalComObject index=255
buffer[0]	hex=0x00 bin=00000000
buffer[1]	hex=0x09 bin=00001001
buffer[2]	hex=0x11 bin=00010001
buffer[3]	hex=0x02 bin=00000010
buffer[4]	hex=0x00 bin=00000000
buffer[5]	hex=0x00 bin=00000000
buffer[6]	hex=0x00 bin=00000000
buffer[7]	hex=0x00 bin=00000000
buffer[8]	hex=0x00 bin=00000000
buffer[9]	hex=0x00 bin=00000000
buffer[10]	hex=0x00 bin=00000000
buffer[11]	hex=0x00 bin=00000000
buffer[12]	hex=0x00 bin=00000000
buffer[13]	hex=0x00 bin=00000000
protocolversion=0x00
msgType=0x09
handleMsgRestart
matching IA
WDT reset NOW

-------------------------
DEBUG! free ram: 1882
KONNEKTING DemoSketch
Initialize KonnektingDevice
15/7/255 = 0x7fff
Manufacturer: 0xdead Device: 0xff Revision: 0x00
numberOfCommObjects: 2
memRead: index=0x00
memRead: data=0x00
_deviceFlags: 00000000
->EEPROM
memRead: index=0x01
memRead: data=0x11
memRead: index=0x02
memRead: data=0x02
memRead: index=0x0a
memRead: data=0x00
memRead: index=0x0b
memRead: data=0x00
memRead: index=0x0c
memRead: data=0x00
ComObj index=0 HI=0x00 LO=0x00 GA=0x0000 setting=0x00 active=0
memRead: index=0x0d
memRead: data=0x5f
memRead: index=0x0e
memRead: data=0x0b
memRead: index=0x0f
memRead: data=0x80
ComObj index=1 HI=0x5f LO=0x0b GA=0x5f0b setting=0x80 active=1
IA: 0x1102
Reset triggered!
Reset attempts: 9
Data available: 1
Reset successful
Init : Normal mode started

Init successful
KnxDevice startup status: 0x00
getParamValue: index=0 _paramTableStartindex=16 skipbytes=0 paremLen=2
memRead: index=0x10
memRead: data=0x05
[... 334 lines cut ...]
Data available: 1
Reset successful
Init : Normal mode started

Reset triggered!
HOT RESET case
Reset attempts: 9
Data available: 1
Reset successful
Init : Normal mode started

Reset triggered!
HOT RESET case
Reset attempts: 9
Data available: 1
Reset successful
Init : Normal mode started

Reset triggered!
HOT RESET case
Reset attempts: 9
Data available: 1
Reset successful
Init : Normal mode started
`

hasselk, 2016-12-12:
ok i tried the same with the Konnekting_communication_test sketch. hardcoded 1/1/1 as Adress.
its going in the same reset loop "HOT RESET case"

`DEBUG! free ram: 1886
Initialize KonnektingDevice
15/7/255 = 0x7fff
Manufacturer: 0xdead Device: 0xff Revision: 0x00
numberOfCommObjects: 2
memRead: index=0x00
memRead: data=0x11
_deviceFlags: 00010001
->EEPROM
memRead: index=0x01
memRead: data=0x11
memRead: index=0x02
memRead: data=0x01
memRead: index=0x0a
memRead: data=0x3f
memRead: index=0x0b
memRead: data=0x07
memRead: index=0x0c
memRead: data=0x80
ComObj index=0 HI=0x3f LO=0x07 GA=0x3f07 setting=0x80 active=1
memRead: index=0x0d
memRead: data=0x3f
memRead: index=0x0e
memRead: data=0x08
memRead: index=0x0f
memRead: data=0x80
ComObj index=1 HI=0x3f LO=0x08 GA=0x3f08 setting=0x80 active=1
IA: 0x1101
Reset triggered!
Reset attempts: 9
Data available: 1
Reset successful
Init : Normal mode started

Init successful
KnxDevice startup status: 0x00
Toggle LED every 2500 ms.
Setup is ready. go to loop...
Actual state: 0
DONE
Reset triggered!
HOT RESET case
Reset attempts: 9
Data available: 1
Reset successful
Init : Normal mode started
`

tuxedo0801, 2016-12-13:
From your log I can see, that the address 1.1.1 is correctly programmed and read out from eeprom. 0x1101 == 1.1.1

On startup (Konnekting.init(...)) the constructor of KnxDevice.cpp is called:
First, it resets the knx transceiver:

https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxDevice.cpp#L76

and after that, Init() is called, which applies the IA 1.1.1

https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxDevice.cpp#L86

So I'm a bit irritated that the IA might cause this issue... 

But the most strange thing is this "HOT RESET case". This *never* happend on our side. 

https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxTpUart.cpp#L102

We will first try to reproduce it and then we will see if further debugging is required.

tuxedo0801, 2016-12-13:
You can try adding debug output just before this line:

https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxDevice.cpp#L493

Looks like the reset is triggered by knx transceiver...

hasselk, 2016-12-14:
i enabled a few debug outputs in KNXTpUart.cpp around `switch (_rx.state) {`
https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxDevice.cpp#L336

i dont know if this is a hint but the Debug output looks like this: main diference is ` incomingByte=0x07` against `incomingByte=0x03`

when it works it looks like this:
`Setup is ready. go to loop...
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x07, readBytesNb=0
Rx: State Indication Received
Actual state: 0
DONE
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x07, readBytesNb=0
Rx: State Indication Received`

when it doesent:
`Setup is ready. go to loop...
Actual state: 0
DONE
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x03, readBytesNb=0
Reset triggered!
HOT RESET case`

tuxedo0801, 2016-12-15:
Same phenomenon with regard to HOT RESET, but independant from individual address:

https://knx-user-forum.de/forum/projektforen/konnekting/941633-universal-up-schnittstelle-1wire-up-mini-gateway-binäreingänge-i²c-spi-digitale-i-os-etc?p=1024261#post1024261

Pls. also check your soldering ...

hasselk, 2016-12-15:
If you have a device like this (its a Devboard HW 1.2), could you try to reproduce the error? Or have you already tryed and you have no troubles?

I will solder a few other devices to this weekend, and check it with them. Because this device realy worked reliable over the last weeks and only has trouble when i use IA with 1 at the end. Il try to record the serial communication between atmega and ncn to see if there is a difference in the initial init.

tuxedo0801, 2016-12-15:
I've not yet tried it explicitly, but I'm, quite sure I already used an IA ending with .1.
@Adminius also reported the he was using an IA ending with .1 successfully. 

Also I don't see why an address with .1 would cause this issues. There's nothing special about the IA on the code. There's absolutely no difference between .1, .0 or .123

hasselk, 2016-12-21:
I retried it with 4 different devices, all behaving the same. Anything works until i use an IA ending with 1.
Right now i think its very unlikely that i soldered 4 devices with the same error in them, especialy when anything else works.
And there are more odd behaviours all related to the lower 4 bits of the low IA:
```
1            -  Hot Reset Case
5            - no send possible BUT recieving goes i see suite data comming in when i try to program
0x0b 11 - Rx: unexpected TPUART_DATA_CONFIRM_SUCCESS received!
0x0f 15   - Rx: unexpected TPUART_DATA_CONFIRM_SUCCESS received! but only with 1.1.15 with 8.5.15 it seems to be ok
```
and again tested on 4 devices, and reproduceable.
Could this be some kind of arduino version incompatibility actualy i use 1.6.12? very strange or i got a complete batch of misbehaving Chips, but that seems unlikly too because of i can reproduce the same errors.
I am very iritated.

Adminius, 2016-12-21:
Hey. I've tested your issue on a new UFI (32U4) with 1.1.241 and 7/7/1 and i had no problems. Have you an Arduino with another uC that yours to verify your issue on another uC?

tuxedo0801, 2016-12-21:
@hasselk
As ist works on our sie, I assume it's a HW issue on your side. Will try to reproduce it with my Kombi Sensor. But I first need to adapt it to beta4 (its still Beta 3 😔).

tuxedo0801, 2016-12-22:
And again:

When the µC starts, first the BCU is reset. This *MUST* work if BCU HW is correctly connected/soldered and KNX cable is connected to it. At this stage, the IA is read out from eeprom and held in a local variable, but NOT YET sent to BCU. So while the initial reset happens, BCU does not know about the IA at all.

According to one of your logs:

```
DEBUG! free ram: 1882
KONNEKTING DemoSketch
Initialize KonnektingDevice
15/7/255 = 0x7fff
Manufacturer: 0xdead Device: 0xff Revision: 0x00
numberOfCommObjects: 2
memRead: index=0x00
memRead: data=0x00
_deviceFlags: 00000000
->EEPROM
memRead: index=0x01
memRead: data=0x11
memRead: index=0x02
memRead: data=0x01
memRead: index=0x0a
memRead: data=0x00
memRead: index=0x0b
memRead: data=0x00
memRead: index=0x0c
memRead: data=0x00
ComObj index=0 HI=0x00 LO=0x00 GA=0x0000 setting=0x00 active=0
memRead: index=0x0d
memRead: data=0x5f
memRead: index=0x0e
memRead: data=0x0b
memRead: index=0x0f
memRead: data=0x80
ComObj index=1 HI=0x5f LO=0x0b GA=0x5f0b setting=0x80 active=1
IA: 0x1101
```
Device is currently starting up. IA has been read from EEPROM.
Startup will continue and initially reset the BCU/KNX Transceiver for communinucation start:
```
Reset triggered!
Reset attempts: 9
Data available: 1
Reset successful
Init : Normal mode started

Init successful
KnxDevice startup status: 0x00
```
Reset was successful. IA has been applied AFTER the reset.
Now we continue setup() with reading the params:
```
getParamValue: index=0 _paramTableStartindex=16 skipbytes=0 paremLen=2
memRead: index=0x10
memRead: data=0x05
 val[0]@16 -> 0x05
memRead: index=0x11
memRead: data=0xdc
 val[1]@17 -> 0xdc
Toggle LED every 1700 ms.
Setup is ready. go to loop...
```
setup() is done, now the loop() follows, which also receives telegrams from BCU/KNX Transceiver:
```
Reset triggered!
HOT RESET case
Reset attempts: 9
Data available: 1
Reset successful
Init : Normal mode started

Reset triggered!
HOT RESET case
Reset attempts: 9
Data available: 1
Reset successful
Init : Normal mode started
```
Here the µC received a data-packet from BCU which indicates a communication problem. The lib will therefore reset the BCU again to "repair" the communication...

And the big question is: What is your HW doing different to ours?! And why does this happen only with IAs ending with .1 ?!

As we right now cannot reproduce the issue, we have no chance to debug it :-(

You have to dig into the source code to check yourself. Maybe try to correlate the occurence of the BCU data packet that causes the hot-reset case with a telegram on the bus? Maybe there is some conflict?

tuxedo0801, 2016-12-22:
Which hardware are you using? HW1.2? Please try to send us a high-resolution image of your PCB. Maybe there is a general soldering issue?!

tuxedo0801, 2016-12-22:
I ported my kombisensor (very similar to HW1.0) to beta4 and programmed IA 7.7.1... 
Works. Will also test 7.7.5, 7.7.11 and 7.7.15 ... stay tuned.

hasselk, 2016-12-22:
That would be realy nice if you find an error on my pcb :)

only IA ending with 1 leads to hot reset, 5 doesent want to send data out, i cant even see comunication on the serial between atmega and ncn with the scope when it should be sending, 11 & 15 seem to work but in the bebug output there is that "Rx: unexpected TPUART_DATA_CONFIRM_SUCCESS received" kind of a crazy collection of errors. and only in the lower 4 bytes.

PCB Top
[https://drive.google.com/open?id=0B8F7Si0gPGnGSHFOcHRYcy10M1E]
PCB Bottom
[https://drive.google.com/open?id=0B8F7Si0gPGnGSjh5TXFYSkNQaGc]

hasselk, 2016-12-22:
Hello Adminius, thanks for confirmation. I have a few nanos here, might have an siemens tpuart. Have to search for it or keep the 32u4 in reset and use the hw1.2 ncn. will report back

Adminius, 2016-12-22:
You can solder out 2x 1K resistors between ncn RX/TX and ProMicro TX/RX. In this case you will have two "Test Points" for nano+ncn Test. But be carefull, nano have 5V logic Level. If have nano with ch340g UART, you have to disconnect this uC.it makes problems with KNX transceivers

hasselk, 2016-12-22:
Thank you @Adminius have not thought  at the resistors that makes it easyer ;) and the 5v would have been my trap ;) will use a levelshifter

hasselk, 2016-12-22:
something came to my mind, is it enough to program the bootloader through arduino ide or do i need to set some special fuses?

Adminius, 2016-12-22:
Arduino IDE bootloader with standard fuses is enough

Adminius, 2016-12-22:
i have new (bad) news :(
UFI (NCN5120+32u4) and AI 1.1.1 have also HOT RESET :( (with 1.1.11 it works)
But, Siemens BCU+32u4 and AI 1.1.1 works.
That means, it is a HW problem... NCN-problem...

tuxedo0801, 2016-12-22:
Question is: Why?

I think we have to investigate into NCN datasheet to find out why the NCN is doing this reset.

Adminius, 2016-12-22:
@hasselk 
pls try to set 
[https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxTpUart.h#L62](url)
to 0xF1 instead of 0x28 and test your boards with IA 1.1.1

tuxedo0801, 2016-12-22:
Ok, Eugen found the root-case (congrats...):

Siemens BCU uses die Control-byte 

0010 1000 ---> 0x28

to tell the KNX Chip that we want to set the IA.

Eugen found out, that the Onsemi NCN5120 is **not compatible** in this case. It requires

1111 0001 --> 0xF1

to tell the KNX Chip the individual address.

We did not know that and sent 0x28 anyhow to the NCN...

0010 10aa is - fot the NCN - the control byte to set an internal register (U_IntRegWr.req), where "aa" is the Register-Address.

As we sent 0010 1000, we sent the register address 0x00 which is the internal WatchDog Register.

The register requires one single byte to set/configure the watchdog. But we sent our IA, which is two byte.

The NCN seems to handle this protocol-fault with a reset, which our lib cannot handle correctly at this stage. 

Solution: We have to tell the Lib which KNX chipset is beeing used so that it can adress the correct control byte for setting the IA

OR

we have to skip setting the IA completely on the KNX chipset.
Telling the KNX chipset is only useful, when you address the KNX device by it's IA. But this *never* happens with our lib. We only use group-address communiction ("broadcast communication") and not direct adressing ("peer-2-peer communication").

I will check how we solve this.

In the meantime, pls replace Line 62 of KnxTpUart with this:

```C++
#define TPUART_SET_ADDR_REQ                  0xF1
```

Sorry for the irritation. This was difficult issue...

tuxedo0801, 2016-12-22:
We tested now on Siemens BCU as well as on NCN:

If you comment out line 233-240 (https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxTpUart.cpp#L233
to https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxTpUart.cpp#L240),
the IA is not set on KNX chipset (which is okay for our project) and the code works on both chipsets.

I will change this accordingly in beta5. 

So for now, comment out the suggested lines, or change 0x28 to 0xF1 ... as you like. works both. But commenting out the suggested lines makes the code run on both chipsets.

hasselk, 2016-12-22:
Thats great to hear, congrats to both of you.
This gives me some peace of mind that im not unable to build a device without error ;) i have tried that it works well. Thanks a lot! The HOT RESET with IA1.1.1 is gone.

BUT the other issues remain:
```
.3   gives a "partly" working Device with occaisional HOT RESETS
.5   NO Data is sent to the BUS, but data can be recieved
.11 .gives "Rx: unexpected TPUART_DATA_CONFIRM_FAILED received!"
      before the change it gave "Rx: unexpected TPUART_DATA_CONFIRM_SUCCESS"
.15  has the same issue as .11 BUT only with IA of 1.1.15 1.2.15 has not this issues
```

@Adminius could you confirm these other ERRORS too?

hasselk, 2016-12-22:
@tuxedo0801 maybe this is a canditate for another initialization parameter or maybe better defines?
like KNX_Type
`    Konnekting.init(KNX_Type, KNX_SERIAL, PROG_BUTTON_PIN, PROG_LED_PIN, MANUFACTURER_ID, DEVICE_ID, REVISION);
`
And KNX_SERIAL_SPEED would be nice too because the NCN can work with 38400, which could save some cpucycles for other stuff?

tuxedo0801, 2016-12-22:
Don't mix up things. Other issue -> new ticket. First we solve this issue: how did you change the code? Did you comment out or change the constant? If you comment out the IA should not affect anything but outgoing write requests.

hasselk, 2016-12-22:
I tried both and both work.

ive seen something that might be interesting too but i dont realy understand,
in [.../KnxTpUart.h#L65](https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxTpUart.h#L65) TPUART_DATA_END_REQ is defined as 0x40, but in the NCN Datasheet page 29 its 47−7F.

And those two dont get mentioned in the datasheet at all, are those general commands? is there a list of all those commands?:
```
TPUART_RX_ACK_SERVICE_ADDRESSED      0x11
TPUART_RX_ACK_SERVICE_NOT_ADDRESSED  0x10
```

tuxedo0801, 2016-12-23:
Of course, both work. But they work differently. If you face problems with IAs, go for the "comment out" solution and report back if problems are gone.

The Data-End-Request is on both chipsets the same. Compare datasheets... Page 29 NCN vs. Page 22 Siemens TPUART

Both use

01LLLLLL 0x47 - 0x7F

where LLLLLL is a value between 7 and 63. 

If you take 

01LLLLLL 

with L always 0, you get 0x40. And in the implementation you will find:

https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxTpUart.cpp#L483

```C++
txByte[0] = TPUART_DATA_END_REQ + _tx.txByteIndex;
```

So the constant is just a base for the calculation during runtime. 

Please stop searching for issues where no issues are. That makes this issue even longer and confusing. 

So, back to topic:

Please report back if commenting out the suggested lines solves your IA problem **completely**. if yes, we can close the discussion, as this bigfix will come with beta5. 

If you have any further issues, please open a new ticket/issue that can be tracked separately.

hasselk, 2016-12-23:
sorry for not being clear enough. Only the HotReset case with .1 is solved completly and the problem with .5 where no data was send to the bus seems to be solved too. Thats good.

BUT and that should be confirmed by someone else too, even if i comment out the setting of the IA there are problems with .3 / .11 / .15

**.3 leads to a partly working device, meaning it seems to send part of the data on the bus but has "HOT RESET CASE" inbetween**

```
Initialize KonnektingDevice
15/7/255 = 0x7fff
Manufacturer: 0xdead Device: 0xff Revision: 0x00
numberOfCommObjects: 2
memRead: index=0x00
memRead: data=0x11
_deviceFlags: 00010001
->EEPROM
memRead: index=0x01
memRead: data=0x11
memRead: index=0x02
memRead: data=0x03
memRead: index=0x0a
memRead: data=0x3f
memRead: index=0x0b
memRead: data=0x07
memRead: index=0x0c
memRead: data=0x80
ComObj index=0 HI=0x3f LO=0x07 GA=0x3f07 setting=0x80 active=1
memRead: index=0x0d
memRead: data=0x3f
memRead: index=0x0e
memRead: data=0x08
memRead: index=0x0f
memRead: data=0x80
ComObj index=1 HI=0x3f LO=0x08 GA=0x3f08 setting=0x80 active=1
IA: 0x1103
Reset triggered!
Reset attempts: 9
Data available: 1
Reset successful
Init : Normal mode started

Init successful
KnxDevice startup status: 0x00
Toggle LED every 1000 ms.
Setup is ready. go to loop...
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x07, readBytesNb=0
Rx: State Indication Received
Actual state: 0
DONE
KNX_WRITE_REQUEST index=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0xbc, readBytesNb=0
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x11, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x03, readBytesNb=1
Reset triggered!
HOT RESET case
Reset attempts: 9
Data available: 1
data not useable: 0x00. Expected: 0x03
Data available: 1
Reset successful
Init : Normal mode started

Rx: Reset Indication Received
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x07, readBytesNb=1
Rx: State Indication Received
Actual state: 1
DONE
KNX_WRITE_REQUEST index=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0xbc, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x11, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x03, readBytesNb=1
[... 292 lines cut ...]
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x0f, readBytesNb=1
Rx: State Indication Received
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x3f, readBytesNb=1
Rx: State Indication Received
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x08, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0xe1, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x00, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x80, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x2b, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x9c, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x11, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x0f, readBytesNb=1
Rx: State Indication Received
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x3f, readBytesNb=1
Rx: State Indication Received
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x08, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0xe1, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x00, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x80, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x2b, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x9c, readBytesNb=1
RX_IDLE_WAITING_FOR_CTRL_FIELD incomingByte=0x0b, readBytesNb=3
Rx: unexpected TPUART_DATA_CONFIRM_FAILED received!
Actual state: 1
```

hasselk, 2016-12-23:
and i forgot, @tuxedo0801 thanks for the explanation of the Data-End-Request!
But im realy not searching for issues where no issues are. I tested this behaviour with 3 Devices all behaving the same. There has to be some miscomunication between the MCU und the NCN.

@Adminius could you please try to reproduce the Errors with .3 / .11 /.15. This would proof thats not only my devices. And it would ease my mind a bit. Thank you upfront.

tuxedo0801, 2016-12-23:
We will investigate... but please keep in mind, that already one line of debug output can kill the whole communication, as it delays execution. Timing is very important for the KNX chipset (400microseconds can change everything!). 

So please don't enable debut output just as you like. The result is inpredictable. 

If you comment out the suggested lines, the KNX chipset is not informed about the IA. ist just sees the IA when you (your microcontroller) tries to write to an comobj. If you just receive telegrams, the IA is totally unused. 

So right now I don't see any connection between the used IA and the issue with .5/.11/.15 ... :-(

hasselk, 2016-12-23:
Im aware of that, i even changed the HTU21 lib so that i can split the requesting of Temp/Humid and the recieving of the data without delays. So that either request or recieve takes less than the 400micros ;)
but thats another story.

Im trying this only with the Konnekting_communication_test to sort out anything from my code. And there only the sending of data by the toggeling of GA1 in the sketch.
Any other adress i try seems to work fine just those three  **.3/.11/.15** behave differently. Thats whats bothering me.

hasselk, 2016-12-23:
Should i open up another issue for the remaining ".3/.11/.15" troubles? Maybe after confirmation from somebody else?

tuxedo0801, 2017-02-22:
If you still face this .3/.11/.15 issue, open a new ticket. As for now, we could not reproduce this issue.

tuxedo0801, 2017-06-02:
any update on this  .3/.11/.15 issue?

Adminius, 2017-06-02:
i've tested 32u4+NCN5120 without any problems. 
1.1.1/.3/.11./.15 works like a charm

Adminius, 2018-05-17:
solved with beta4a

closed 2018-05-17 by Adminius