KONNEKTING/KonnektingDeviceLibrary #42: Check and maybe improve timing/handling of Task() call.

tuxedo0801, 2017-04-21:
Each and every Knx.Task() call receives/processes just a single byte. 
So it needs to be called several time to get a complete telegram from bus. 

The end of a telegram is indicated by a 2ms "silence".

It's documented that one should call Knx.Task() every ~400µs to not miss a telegram or do not corrupt it.
Beside that, every Arduino has a receive buffer, which gives us ~33ms before the buffer overflows and we miss technically bytes from UART. 

But that does not allow us to "block" the loop() with ~30ms, because we than corrupt a telegram, because we force a "silence" where there is no silence.

One should try to measure how long we can really block the loop() without missing a telegram that we are interested in.

This issue can be solved by using a timer that calls Knx.Task() fast enough. But one still has to ensure, that the timer is running an not blocked by other stuff.

For doing tests, it would be good to

* run an isolated test (where no other KNX devices jump in)
* have a well-defined load on the isolated bus (almost 100%)
* have a test-telegram that the arduino is interested in, f.i. every 1sec
* have a delay in loop() that blocks the loop (test with different, increasing values)
* check at which delay we miss the important telegram that we should receiver every 1sec

Adminius, 2017-06-02:
just my 2cents:
Arduino has a "stream" buffer that can store up to 64bytes, that is correct.
But, our library will read only one byte at once... so it can detect a "gap"=EndoOfTelegram (2-2.5)ms.
If we will store more then one byte, it can't detect this gap any more...
In worst case we have 2 or more telegrams in buffer, but we don't know wicht byte belongs to wich telegram => currupted data.

I would say, we can try this way:
https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxDevice.cpp#L116
it should be an if statement added at the beginning of Task() function:
```
If (_serial.available >0 || txDataAvailable)
{
    serialActionAvailable = true;
}
```
the other part in a Task-function should be placed in a while-loop:
```
while(serialActionAvailable)
{
...
}
```

serialActionAvailable will false, if there are no rx/tx actions are available (e.g. gap was detected)

this sollution blocks loop(), but receives all telegramms correctly.

tuxedo0801, 2017-06-02:
You cannot "work around" the receive-buffer. It's always there, builtin.
Read always happen on the buffer. If you read very fast and often, it's unlikely that more than 1byte is in the buffer. If you read not that often, it's likely that there is more than 1 byte in the buffer.

Your suggested solution is incomplete: When do you set the serialActionAvailable flag back to false?

If the buffer really fills up that fr, so that we cannot detect the 2,5ms-enf-of-message indicator, message cannot be read. That's right.
But right now, I do not have any idea yet how to solve this.

Adminius, 2017-06-02:
> Maybe call task() again while we are in task() unless end of message is detected?

that is a point i've described above

tuxedo0801, 2017-06-06:
Maybe obsolete due to #43

Adminius, 2018-08-12:
solved with #60

closed 2018-08-12 by Adminius