KONNEKTING/KonnektingDeviceLibrary #43: Rework KNX telegram reception

tuxedo0801, 2017-06-06:
**Problem:**

2,5ms "silence" to detect end of telegram --> conflicts with reading data from rx-buffer --> The buffer kills this silence!

**Solution:**

According to datasheet:

http://www.hqs.sbt.siemens.com/cps_product_data/gamma-b2b/TPUART2_technical-data.pdf

> The host controller which is connected to the TP-UART-IC needs either
to detect a receive time-out of 2 to 2,5 ms to detect an end of Packet **or check the CRC16-CCITT**. 

DKA TPUart library is reading 6 bytes header, followed by n bytes payload + CRC. And THEN the end of packet is reached. DKA has the payload in byte no. 6 (index 5)

The payload is encoded in the routing-byte:
https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxTelegram.h#L44
And we already can read this:

https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxTelegram.h#L106
https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/KnxTelegram.h#L268

So, we should think about reworking the telegram receiption handler to:

1. read 6 bytes header first/upfront
2. get the payload-length
3. read the whole payload (blocking read!)
4. read crc and what is followed
5. --> telegram received completely

**Conclusion:**

When not relying on the 2,5ms silence to detect end of packet, we can totally rely on the 64byte RX buffer on the serial uart. That gives us enough time to do other blocking/time consuming stuff (like i2c, ...). And then it does not matter whether we read a complete telegramm "en block" or read it byte by byte.

**Open Question:**

Do we need to set a speciall register to enable CRC16-CCIT? Does it differ from Siemens TPUART/2 to NCN5120 etc?

Adminius, 2017-06-06:
> Do we need to set a speciall register to enable CRC16-CCIT? Does it differ from Siemens TPUART/2 to NCN5120 etc?

it seams to be different:
TPUART: 0x25
Elmos E981.x3: 0x25
NCN5120: 0x1E ? (page 29, NCN5120 datasheet)

So, we have to find a possibility to automatically detect a tranceiver type or we have to set it manualy in a init function. (#41)

tuxedo0801, 2017-06-06:
Question is:
Do we actually NEED to set this?
Right now we also have checksum implementation, without setting any special register. Even DKA is using the checksum and does not set anything (at least I did not find anything).
So one has to check if this IS required or not.

Adminius, 2017-06-07:
if we need it:
NCN has as Service Register, that is not available on TPUART and E981.x3.
In this case we can send B00001101 to NCN an and the answer is 2 bytes, first of it is a ACK: B01001011
Tpuart doesn't send any ACK in this case.

tuxedo0801, 2017-06-07:
How does 2nd byte look like?

Adminius, 2017-06-07:
the 2nd byte looks like: B11111011. (NCN5120 Datasheet, page 34.)
[image]

tuxedo0801, 2018-01-11:
Check difference between siemens tpuart and ncn in default mode (no explicit crc check enabled).
Check if crc bytes (if there are any, DKA states that there is one, see: https://bitbucket.org/dka/arduino-tpuart/src/e07838b568823559e3bd09625ac8004f04fdee94/KnxTpUart/KnxTpUart.cpp?at=default&fileviewer=file-view-default#KnxTpUart.cpp-121) are indeed a kind of crc and if CRC16-CCIT is only "additional on request be setting the config register accordingly".

tuxedo0801, 2018-01-11:
Last byte of telegram is CRC checksum, which is XOR sum of each byte of telegram (except crc byte).

helpful links: 
https://knx-user-forum.de/forum/öffentlicher-bereich/knx-eib-forum/diy-do-it-yourself/33016-arduino-am-knx?p=1141860#post1141860
http://www.see-solutions.de/sonstiges/KNX%20Twisted%20Pair%20Protokollbeschreibung.pdf

So we will use this instead of 2 byte CRC16-CCIT

shacal, 2018-02-07:
CRC is telegram CRC, nothing prevents TPUART sending bytes, then middle of header stop, send state because low voltage happens etc. Yes, we have save pin for that reason (also should happen in irq and not be blocked). To fix this issue, I have used small avr to actually read bit by bit and when message received with 2ms delay mark it as full package and resend it to master mcu that has proper message capsulate so buffer can be large and tasks take more time. Making all this happen in MCU is tricky, as if you have any tasking/threads context switch can take more than 20ms.
Arduino Serial buffer is bad for Siemens TPUART.

So, having an library that is drop in easy to use, might be mighty task if you need to prevent irq and wanna share i2c or spi. As TPUART was never meant to be directly connected to multi task device.

Adminius, 2018-08-12:
i've implemented new reception. #60 
now we know after 6th byte how long a telegram will be.
so we can read telegrams from a RX buffer.
But one Problem is not solved: after a full telegram we should send a ACK within 1.35ms.
If we read the whole telegram from buffer we don't know a when the last byte was received. 
that means we cannot send ACK within 1.35ms after last byte...

shacal, 2018-08-13:
ACK is OK to send when you know address and length that is valid (7 <= 16 == std frame).
As study in https://pdfs.semanticscholar.org/de58/3b5ff76705168a6b2de27864c560678e502a.pdf shows, 1.32ms in not strict and using Arduino and buffered serial (not interrupts) you are already late anyway.

tuxedo0801, 2019-10-24:
According to @Adminius :

We're sending the ACK, but not within/after 1.35ms (due to rx buffer of arduino).
Right now we also face no further issue with this, so the ticket can be closed for now.

closed 2019-10-24 by tuxedo0801