KONNEKTING/KonnektingDeviceLibrary #47: INIT-read request should be done only one time

Adminius, 2018-01-18:
AS IS: infinity loop with init-requests
TO BE: only one init read request. if no answer -> undefined value

tuxedo0801, 2018-01-18:
How do you define "undefined value"? 0x00? 0xFF? Both can have different meanings.

Adminius, 2018-01-18:
int tmp = 0; is defined value
int tem; is undefined value

tuxedo0801, 2018-01-18:
But 0 could also be a valid value... You would need an extra flag to "flag" this value as "undefined". But what does you KO answer if one reads it's value? You "have to" reply an value. There's no option to tell in the answer that the value is undefined.

tuxedo0801, 2018-06-08:
```
int tmp = 0; is defined value
int tem; is undefined value <---- How do you "detect" that a variable is "not initialized"?
```

tuxedo0801, 2018-06-08:
Just tested:

If you define a global variable like this:

```
int foo; 
```

and do a console print with the value if this variable, you get 0. 

So there is no way to detect uninitialized values. 

The only thing we could do:

create a delay in the "read" loop for init-comobjs to reduce the speed a bit and maybe do it only for a specific amount of time, let's say 20sec or 1min to give the device a chance to startup if the queried device is not reachable at all.

Adminius, 2018-06-08:
another way:
INIT function should be called only on device start.
after one init-read-execution we set a global comObjInit = true flag.
That means, that init-read will never be called anymore (until device reset)
if no one "answers" knxEvents will not be called -> FW developer should take care of default values

tuxedo0801, 2018-06-08:
So you suggest to just read once? Why not read f.i. at most 10 times with 1sec delay? That would enable the device to get an init value, even if the bus is temporarily overloaded?

Maybe we should also read the knx spec. Or we try an existing device, configure an comobj for init-read and apply an adress that is not available and see what happens.

Adminius, 2018-06-09:
Yes, just once... 
My "not konnekting" device do it just once too...

tuxedo0801, 2018-06-10:
ok, then we do the same.

tuxedo0801, 2018-07-30:
merged, close for now.

closed 2018-07-30 by tuxedo0801