KONNEKTING/KonnektingDeviceLibrary #49: make reboot() public

SirSydom, 2018-03-14:
to enable a reboot from sketch

tuxedo0801, 2018-03-14:
Would be possible, sure. But just to get an idea:
Why do you want to reboot your device from sketch?

SirSydom, 2018-03-15:
I have a temperature monitoring which sets the device into an emergency mode if temp is to high.
After the alarm the device is resetted, which is the simplest and cleanest way to go back to normal operation.

tuxedo0801, 2018-03-15:
you're monitoring the device's temperature or an external temperature of something else?

I don't get it why it's possible by you implementation to set emergency mode and also do an reset, but not that easy to unset the emergency mode.

What's so special with this emergency mode?

SirSydom, 2018-03-15:
internal temp. with an onewire sensor.
There ist nothing special, it is not impossible, but it is the simpliest way regarding developing time, code complexity, and flash consumption.

Adminius, 2018-06-10:
@SirSydom any new? i this feature still required?

SirSydom, 2018-06-11:
yes I need it.

tuxedo0801, 2018-06-11:
The use-case of resetting the controller is very rare and very special. We would not open the API and make the reset() function "public", as this will give the feeling that it's quite okay and "good code" to do reset from the sketch.

If you really insist in using the reset-mechanism, I'll suggest to copy&paste the code to do the reset in your special case.

closed 2018-06-11 by tuxedo0801