KONNEKTING/KonnektingDeviceLibrary #65: Busware USB/KNX interface can not work?

Jason0416, 2019-08-15:


tuxedo0801, 2019-08-15:
1st of all: please provide more information.

2nd:

We did not test for busware interface, but in principle, it's just a usb-rs232 to KNX Transceiver interface, so it should work. But question is: why do you want to put a USB-KNX Interface to an arduino? Remember: You posted your question on the KonnektingDeviceLibrary repository. 

If you want to use the interface in combination with the KONNEKTING Suite, then it should work, if the appropriate firmware is on the interfacce. But as I said: We did not test this and without any further information, we cannot give any advise to to get it to work.

br,
Alex

Jason0416, 2019-08-26:
OK，Move to KONNEKTING suite.

closed 2019-08-26 by Jason0416